circuitnet/CircuitNet #10: Question about label

Doctor-James, 2022-12-07:
I would like to ask about the label in the congestion prediction and DRV prediction problem. The value of each pixel in the label in the dataset is between 0 and 1. What is the actual physical meaning of this? I see that some other articles define congestion prediction as a classification problem, and the label value is 1 or 0. Is there also a threshold so that the pixel value above the threshold represents congestion? The label in DRV prediction is a similar problem

apri0426, 2022-12-10:
Yes, some articles define these tasks as a classification problem, but the model is trained in a regression manner, and the prediction is turned into binary with a threshold, then we plot ROC curve and PR curve to evaluate the result.

Doctor-James, 2022-12-12:
How is this threshold set, or is it also a hyperparameter？In addition, do the values in the label have actual physical meaning？

apri0426, 2022-12-13:
Of cource the values have physical meaning, in DRC task, the values are the number of DRC violations counted in each tile, and in IR drop task, the values are the IR drop values in each tile. 

The threshold is set according to actual needs. For DRC, we want to find areas with most DRC, and let EDA tools to optimize these areas. Similarly, for IR drop, we have some specifications that the max IR drop should be limited under specfic percent, like 5-10%, then we set this as threshold.

Doctor-James, 2022-12-14:
Thank you for your answer. 

Actually, my question is about congestion prediction. What is the physical meaning of congestion value? The congestion value in LHNN is described as follows: “set the congestion maps as a binary indicator according to whether the horizontal/vertical routing demand of the G-cell  exceeds the circuit's capacity”.. Here is converting the congestion prediction problem to a classification problem, and to align with it, I also want to convert the label of congestion in CircuitNet to binary. So I want to figure out what the physical meaning of this label is, is there a threshold that converts it to binary? 

I am new to this field, so my basic knowledge is relatively weak. Thank you for your patient answer

closed 2023-02-07 by apri0426

zhengsizuo, 2023-05-30:
In congestion prediction task, labels are generated from overflow files, and after resizing and normalizing, the values are between 0 and 1. If I want to plot ROC curve, how should I decide the threshold label to convert labels into binary?