circuitnet/CircuitNet #11: Struggling to align instance placement and node attributes

choltz95, 2023-01-20:
Hi! Maybe a dumb question, but the docs mention that there is one-to-one correspondence between the node name from the node attribute array above and the keys in the instance placement dictionary.

However, when I check examples, it does not look one to one. For example if I run

```
nodeattr = np.load('node_attr/RISCY-a-1-c2_node_attr.npy', allow_pickle=True)
instance_placement = np.load('instance_placement/510-RISCY-a-1-c20-u0.7-m3-p2-f0',allow_pickle=True).item()
len(nodeattr[1]), len(instance_placement)
```

I get `(53587, 50052)`, different sizes and I see that different instance placements for the same design also have different number of standard cell placements. Do you have any guidance for this or an example of the node attribute alignments with instance placement?

apri0426, 2023-01-24:
Yeah, the node attribute always has more values than instance placement, because the node attribute array is directly from netlist and the instance placement array is from DEF after placement, which has been processed and reduced by Innovus. 
In our own experiment, we take the intersection of two arrays and this part has one-to-one correspondence.

choltz95, 2023-01-24:
That's great- thanks! I am doing the same thing: taking the intersection. I just wanted to verify to make sure it's standard.

closed 2023-01-24 by choltz95

Doctor-James, 2023-05-07:
Hello, I would like to ask for advice：I can understand “the node attribute always has more values than instance placement”, but I found that some of the cells in the instance placement are not found in the node attribute, which means they are placed but not connected to any other cells?