circuitnet/CircuitNet #13: About using circuitnet.lef

linjingui, 2023-03-06:
Hello, I have the following two questions to ask you. 
First of all, I want to ask if your def file is  obtained after the placement of the whole process runs? 
Secondly, I want to view the placement results by importing your circuitnet.lef and def files into EDA business tools, but I found that there was an error when I first imported the circuitnet.ef file. Do you have any suggestions?

Error occurred when I imported the lef file：
Loading LEF file /mnt/hgfs/mkshare/circuitnet.lef ...
**ERROR: (IMPLF-53):	The layer 'M2' referenced in pin 'Z' in macro 'AN2_0010' is not found in the database. A layer must be defined in the LEF technology LAYER section before it can be referenced from a macro. Review the LEF files specified in the init_lef_file variable to see if the layer does not exist or is specified after the one that defines the macro.
Type 'man IMPLF-53' for more detail.
**ERROR: (IMPLF-3):	Error found when processing LEF file '/mnt/hgfs/mkshare/circuitnet.lef'. The subsequent file content is ignored. Refer to error messages above for details. Fix the errors, and restart tool again.
Type 'man IMPLF-3' for more detail.
**ERROR: (IMPLF-26):	No technology information is defined in the first LEF file.
Please rearrange the LEF file order and make sure the technology LEF file is the
first one, exit and restart tool.
**ERROR: (IMPLF-26):	No technology information is defined in the first LEF file.
Please rearrange the LEF file order and make sure the technology LEF file is the
first one, exit and restart tool.

apri0426, 2023-03-13:
Sorry for my late answer. The layer and via definition are not included in the current release of LEF due to NDA, and that's the reason for the errors. We are working on sanitizing them and the new LEF and DEF will be released in a few days, and the new DEF is dumped after routing. I will inform you in this thread when we are done.

linjingui, 2023-03-13:
That's fine. Thank you!

linjingui, 2023-03-22:
Hello, I found that you uploaded a new LEF file today, I re-downloaded it and put it into the commercial tool to read it when I reported an error, is it not completely replenished the LEF file content?Secondly, I would like to ask if the current def file has been modified?Please guide me, thank you！
[image]

apri0426, 2023-03-22:
Thank you for your report! I just upload a new LEF, please take a try. 
Yes, the LEF has been modified this time so the new DEF is also modified accordingly. 
We normalized coordinates when sanitizing. It is revertable with supplemental information which can be obtained upon email request.

linjingui, 2023-03-23:
First of all, thank you very much for your reply, and secondly, I have three questions, I hope you will not be bored, first: Regarding what you said about requesting this change by email, I would like to ask how to communicate by email, am I here to send you my email account or others? Second: Regarding your newly uploaded ispd2015 features, do you plan to upload the corresponding DEF file? Third: Is the newly uploaded DEF file of Circuitnet placement or global routing?

linjingui, 2023-03-23:
Hello, I just tried to read the LEF and DEF files, the LEF file did not report an error, but the following error occurred when reading the def file 1-RISCY-a-1-c2-u0.7-m1-p1-f0.def（For convenience, I changed its name to 1.def）
[image]

apri0426, 2023-03-23:
> Hello, I just tried to read the LEF and DEF files, the LEF file did not report an error, but the following error occurred when reading the def file 1-RISCY-a-1-c2-u0.7-m1-p1-f0.def（For convenience, I changed its name to 1.def） [image]

It seems that your tool do not support halo, you may need to remove the halo from DEF to proceed. The halo is defined for macros which are all in the first several lines of COMPONENTS part in the DEF, and you can remove them by removing the line starts with "+ HALO". 

The missing tech-via error is quite strange, as these via definitions have been included in the DEF. 

You can try to remove the halo first and see if there are more errors.

apri0426, 2023-03-23:
> First of all, thank you very much for your reply, and secondly, I have three questions, I hope you will not be bored, first: Regarding what you said about requesting this change by email, I would [...]

1. My email is zhuominchai@whu.edu.cn. 
2. Yes, we can upload DEF for ispd15.
3. The DEF is writ after detailed routing.

apri0426, 2023-05-29:
If you want to load LEF DEF with openroad, you need to remove HALO in components part and virtual net in nets part, and add NDR via definition in LEF. We have prepared a LEF with NDR via definition, and you can contact me with email (zhuominchai@whu.edu.cn) to request it.

closed 2023-05-29 by apri0426