circuitnet/CircuitNet #14: Problems with Gate-level Netlist

Doctor-James, 2023-03-10:
I'd like to ask some questions about Gate-level Netlist，I looked at the contents of node_attr，it's a two-dimensional array，The first dimension starts with "pulpino_top", followed by some node names，the second dimension starts with "module", followed by some module names. Does that mean nodes which have the same module name are connected to each other?

apri0426, 2023-03-11:
The name "pulpino_top" and "module " are actually unexpected in this array, they are removed from the upload today (2023/3/11), thank you for pointing it out.

node_attr is the feature extracted from netlist to construct graph, and there is a page about its characteristics [https://circuitnet.github.io/feature/graph.html](url). 
Nodes with same module name doesn't mean they are connected. Nodes are connected when they belong to the same net.

For the netlist itself, checkout the flatten_netlist.tar.gz or hier_netlist.tar.gz.

Doctor-James, 2023-03-13:
flatten_netlist.tar.gz or hier_netlist.tar.gz. The file inside was a verilog file, and it was so complicated that I was confused. Is there an easy way or script to get information about connections between nodes? (used as input to GNN). Thank you for your patience

xiaoqiang88888, 2024-10-30:
非常抱歉打扰您一下，就是存在pin 属于一个list net的情况如何处理，例如：['RTSEL' list([41231, 41231]) 39721]

Doctor-James, 2024-10-30:
> 非常抱歉打扰您一下，就是存在pin 属于一个list net的情况如何处理，例如：['RTSEL' list([41231, 41231]) 39721]

你好，我们收到了你的邮件，关于一个pin属于多个net的情况是很正常的，你可以以同一个net为主体，然后去找他包含的所有pin，从而找到包含的所有node，另外，我们的circuitformer由于是类似点云的，所以不需要关注各个node之间的拓扑链接情况，只需要关注几何位置

xiaoqiang88888, 2024-10-30:
您好，冒昧打扰您一下。请问是相当于把这个list拆开处理吗？把list里的net 逐个取出来进而根据您提供的建议“以同一个net为主体，然后去找他包含的所有pin，从而找到包含的所有node”。

  冒昧的问一下您，请问是否需要考虑edge的权重？比如之前的数据中，39721这个node和41231这个net连接了两次。

  如果是下面这么个复杂的情况，同一个net连接了和node 连接了32次。

xiaoqiang88888, 2024-10-30:
您好，冒昧打扰您一下。请问是相当于把这个list拆开处理吗？把list里的net 逐个取出来进而根据您提供的建议“以同一个net为主体，然后去找他包含的所有pin，从而找到包含的所有node”。

  冒昧的问一下您，请问是否需要考虑edge的权重？比如之前的数据中，39721这个node和41231这个net连接了两次。

  如果是下面这么个复杂的情况，同一个net连接了和node 连接了32次。

  

  老师，您好，顺便再补充一个问题，如果如果遍历出来的某个net里只有一个node节点，是不是可以忽略呢，