circuitnet/CircuitNet #15: I have some questions about data set production

gjhhust, 2023-03-21:
Hello! I recently read your paper and code, also want to make your own dataset training attempt, but I found that your dataset is used innovel software analysis reports into NPY files? I'd like to talk about some of the benchmarks of the ISPD contest being converted into datasets. At present, in addition to the notes and https://www.ispd.cc/contests/11/other_files/benchmark_format.pdf provided by the benchmarks, after the benchmarks I used some open source software to generate the global placement file PL (containing all pin lower left coordinates, Level 1) , as well as the global routing file GR (including all pin connections, multi-layer wiring, 0-9 layers) , I observed that your dataset is a single layer of files and wiring, can you make a dataset? Thank you for your reply

apri0426, 2023-03-21:
Our dataset is for machine learning applications, and what we provide are the "hand-crafted" features like cell density and RUDY which maps the actual cells and wires into single layer numpy array, which can be directly fed into the model, instead of detailed placement and routing results. 

What you get from ispd11 and open-source software may be in bookshelf format, a commonly used format for backend design by academics. And what we used in our flow is the LEF&DEF format, and they also contain all the P&R information and we also released them.

apri0426, 2023-03-21:
We haven't released the code for making datasets. But we are preparing a tutorial for this, it should be released together with the code some time later, maybe in May.

The multi-layer in routing is not considered, as most things (feature extraction, prediction, etc.) we do are in the placement stage, and routing is not available. Instead, we mainly consider cell locations and pin locations.

You can take a look at some papers to get more background, like this: Z. Xie, et al. “RouteNet: Routability prediction for mixed-size designs using convolutional neural network,” in ICCAD 2018. 1–8.

gjhhust, 2023-03-22:
Thanks for the reply!!! I have read this article while using your CircuitNet and understand the stages of each feature extraction, RUDY and cell after detail placement combined with globally placed PL file to extract that feature, GR congestion is extracted from the information of global wiring obtained after global routing, the GR information I got so far I checked is found to contain (multi-layer connectivity) (each layer has different horizontal and vertical capacities), I observed the structure of your dataset containing both vertical and horizontal overflow, I think there are two ways to handle this in multi-layer wiring.
- The horizontal overflow of the board can be obtained by superimposing all the horizontal overflows in the same tile (xy is the same, z is different), and the vertical overflow is the same
- Separate the entire multilayer, that is, if the board wiring contains 9 layers, then I can get 9 data sets, each is its overflow, and because the pin itself is in a different layer, all the pin location in each data is also different

Finally, 1-RISCY-a-1-c2-u0.7-m1-p1-f0, this is a dataset naming, I want to ask the naming principle, c2 and u0.7 I do not quite understand, m1 is the m1 layer? p1 is pin + id, f0 I also do not quite understand

apri0426, 2023-03-22:
What global routing do is to assign wires into global routing cells (Gcells) to facilitate the search in detailed routing, and each global routing cell has technology-defined capacity on each layer in 2 directions. So when more wires are assigned into 1 Gcell than its layer capacity, the Gcell is congested. Finally, the congestion map is merged into single layer based on the partition of Gcells, and usually commercial tools can directly generate it after detailed routing. 
In a word, we directly get congestion map from Innovus. So you can read the doc of the tool you used to find how to report congestion, or just use another tool like OpenRoad that can do it.

Here is our naming convention https://circuitnet.github.io/feature/properties.html.

RandomFeng, 2023-12-18:
> We haven't released the code for making datasets. But we are preparing a tutorial for this, it should be released together with the code some time later, maybe in May.
>
> The multi-layer in routing is [...]

Have you released the code of generating datasets now?

closed 2024-03-04 by apri0426