circuitnet/CircuitNet #20: RUDY map pre-process

zhengsizuo, 2023-05-26:
Are your RUDY map of RISC-V dataset pre-processed by smoothen? But the ISPD2015 dataset provided by you are not?
Because the data distribution is very different.

apri0426, 2023-05-29:
No, they are processed in the same way. 
If you are comparing RISC-V with superblue from ISPD2015 after resizing to the same size, they should be very different due to difference in design scale and technology nodes.

zhengsizuo, 2023-05-30:
yes, it seems that the bin size of RISC-V is 1.5μm×1.5μm and the bin size of ISPD2015 is 3.0μm×3.0μm. After resizing the superblue into 256*256, I tested the pre-trained model from RISC-V on superblue, the performance is not good. So is it reasonable?

apri0426, 2023-05-31:
Yes, the diversity of our dataset is not yet enough to cover superblue designs, so fine-tuning or retraining is necessary to reach good performance.

closed 2024-03-04 by apri0426