circuitnet/CircuitNet #22: python process_data.py

GuoFeng97, 2023-09-16:
Traceback (most recent call last):
  File "/data/gf/anaconda3/envs/circuit-gnn/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/data/gf/anaconda3/envs/circuit-gnn/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/data/gf/CircuitNet-icisc_2023/CircuitNet-icisc_2023/feature_extraction/process_data.py", line 25, in read
    process_log.get_IR_drop_features()
  File "/data/gf/CircuitNet-icisc_2023/CircuitNet-icisc_2023/feature_extraction/src/read.py", line 61, in get_IR_drop_features
    x, y = i.split(',')
AttributeError: 'float' object has no attribute 'split'
Is there any problem？

apri0426, 2023-09-20:
It seems that you are not on the latest commit, please use git pull to update and try again.

GuoFeng97, 2023-10-01:
Thanks for your  replly! But even I pull the latest commit for the latest data, the problem is still here.
[image]
[image]

apri0426, 2023-10-09:
You can add 'print(i)' before this line and send me the output. 
Anyway, this code works fine in my environment, you may also provide your environment information to help me to find the problem.

closed 2023-11-16 by GuoFeng97