circuitnet/CircuitNet #25: delay_prediction中build_graph.py问题

NeverLuck2001, 2023-09-22:
请问使用post_place中的数据运行build_graph.py时，报错
```
Process Process-4:
Traceback (most recent call last):
  File "/home/xx/.conda/envs/env1/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/xx/.conda/envs/env1/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "build_graph.py", line 48, in build_graph
    g.ndata['nf'] = torch.tensor([pin_positions[nodes[i.item()].replace('\\','')][0:4] for i in g.nodes()]).type(torch.float32)
  File "build_graph.py", line 48, in <listcomp>
    g.ndata['nf'] = torch.tensor([pin_positions[nodes[i.item()].replace('\\','')][0:4] for i in g.nodes()]).type(torch.float32)
KeyError: 'core_region_i/CORE.RISCV_CORE/id_stage_i/U148/A1
```
类似报错很多，均是KeyError
KeyError: 'core_region_i/CORE.RISCV_CORE/ex_block_i/alu_i/FE_OFC299_div_en_ex/I'
KeyError: 'axi_interconnect_i/axi_node_i/_RESP_BLOCK_GEN[0].RESP_BLOCK/AW_ADDR_DEC/U31/A1'
...
而build_graph.py在post_cts和post_route上能正确运行没有上述错误。
补充：在post_route上运行时发现缺少几个pin_positions，修改代码跳过后能运行。

apri0426, 2023-09-22:
收到，之前我们主要在post_route上测试的，我来找一下问题

apri0426, 2023-10-09:
这个问题来自于innvous的opt_design。这个功能会对cell进行替换以优化timing，替换主要集中在cts阶段。而我们的instance_placement是从route之后提取的，经过了opt_design，所以跟post_place阶段的cell对不上。
这个问题其实不太好解决，post_place的feature可能不太能用，除非我们再在place阶段提instance_placement。看你的描述你应该不在乎是哪个阶段？建议先用post_route和post_cts。

NeverLuck2001, 2023-10-09:
了解了，感谢！

closed 2023-10-09 by NeverLuck2001