circuitnet/CircuitNet #27: question about downloading datas

pjy0422, 2023-11-07:
hi, I want to download your dataset files with google drive on my linux server.
I know you provided second option as baidu netdisk, but I don't know how to use baidu.
So I tried downloading with google drive:
gdown --folder https://drive.google.com/drive/folders/1Xp2y29Le6Doo3meKhTZClVwxG_7z2QuF?usp=sharing
(this is your CircuitNet-N28)

But it failed with below messages:
Sorry, you can't view or download this file at this time.
Too many users have viewed or downloaded this file recently. Please try accessing the file again later. If the file you are trying to access is particularly large or is shared with many people, it may take up to 24 hours to be able to view or download the file. If you still can't access a file after 24 hours, contact your domain administrator.

I googled this, but couldn't find a solution. 
Could you tell me how to download your files with google drive in Ubuntu terminal?

apri0426, 2023-11-14:
This may be an issue associated with gdown. As suggested by the message, the file is particularly large, so gdown does not work well with it and you may have to download it manually on a web browser.

pjy0422, 2023-11-15:
Thanks for the answer, I'll close the issue.

closed 2023-11-15 by pjy0422