circuitnet/CircuitNet #33: Fail to decompress IR-drop data in CircuitNet-N14, in files ended with .gz00/01/02......

RobertCHU01, 2024-04-11:
I try to run the "decompress_IR_drop.py" but always get the error message "unexpected end of file".

Even if I try to run the gzip commands directly in terminal, I either get "unexpected end of file, uncompress failed" or "unknown suffix -- ignored ".

I've also updated my gzip and try to open the files in graphic UI, but Mac shows "no application set to open the document".
I would really appreciate it if you can tell me where is the error.

[image]

apri0426, 2024-04-14:
You are putting the "cell_edges" files in the wrong directory, move them away. 
Then update the "decompress_IR_drop.py" and try again

RobertCHU01, 2024-04-17:
Thank you for you advice. I've removed all those unrelated files. But when I try "python decompress_IR_drop.py". I get "gzip: ../IR_drop_features/power_t.tar.gz: unexpected end of file". 

Then I try to use the following commands, which untar directly. I get error message "x power_t/RISCY-FPU_freq_50_mp_3_fpu_50_fpa_1.5_p_5_fi_ar.npy: truncated gzip input":

cat power_t.tar.gz*>power_t.tar.gz
tar xzvf power_t.tar.gz 

These things happen when I am processing data in N14 on my MacBook with M1.

apri0426, 2024-05-07:
Sorry for my late answer. Some of the power_t.tar.gz files may not be downloaded successfully, since they are pretty large. You can check the local file size that they match the ones on Google Drive, otherwise, you may need to download them again.

RobertCHU01, 2024-05-15:
Thanks a lot

closed 2024-05-15 by RobertCHU01