circuitnet/CircuitNet #36: For the issue of mismatched files in the routing features section of N14

lichengjin10, 2024-05-24:
When I run python generate_training_set.py for CN-14,There are 10 files in the label that do not match.
[image]

I looked at the question commented by luoxiaotian521 on April 4th, and you mentioned the CSV file for N14. I generated a name_list using the file name provided in the CVS file to run generate-training_set.py, but there is still an issue with the file not being present.

[image]
so,how can i solve these questions

apri0426, 2024-05-26:
The issue originates from a mismatch in the number of files for each feature in N14. I have updated generate_training_set.py to avoid this issue for now, and we will update the features to match their numbers soon.

lichengjin10, 2024-05-26:
Okay, thank you for your reply. Looking forward to your update.

closed 2024-05-26 by lichengjin10