circuitnet/CircuitNet #38: FeatureExtract different from

zhsky2017, 2024-06-24:
When I used the code in the feature_extraction folder to extract features, I found that the npy features generated by the script were inconsistent with the routability_features_decompressed features downloaded from googledrive. For example, I used 10129-zero-riscy-b-3-c5-u0.9-m1- The two defs corresponding to the p6-f0 design (place stage and route stage) and circuitnet.lef generate cell_density feat. They are inconsistent with the downloaded cell_density features and there are a lot of strange noises. The two defs I used can be found in the compressed package. To obtain, the code uses the feature extraction in the repo and only modifies the path.
[place&route-DEF.zip](https://github.com/user-attachments/files/15954153/place.route-DEF.zip)

[image]
[image]

apri0426, 2024-07-09:
We have identified the issue and will address it by uploading new DEF files in a few days.

apri0426, 2024-11-09:
The new DEF files have been uploaded.

closed 2024-11-09 by apri0426