circuitnet/CircuitNet #42: N14数据集问题

zhu-jinbo, 2024-11-07:
Hello, have you solved this problem?
`https://github.com/circuitnet/CircuitNet/issues/36`
Why did I still encounter the same problem？
[image]

apri0426, 2024-11-09:
Please pull the repo, or update the csv here: https://github.com/circuitnet/CircuitNet/blob/main/routability_ir_drop_prediction/files/train_N14.csv.
This csv does not contain the names in your errors and should work fine.

zhu-jinbo, 2024-11-09:
But we encountered a file missing issue when executing a preprocessing script to generate a training set for the corresponding task, `python generate_training_stet.cpy -- task IR_rop -- data_path [path_to-decompressible d_dataset] -- savepath [path_to-usave_output]`, Will the train_14.csv file you proposed work at this time? And when we tried to update train_14.csv using your method, we also encountered the same error.

apri0426, 2024-11-12:
I update the generate_training_set.py script, and it should be working now.

zhu-jinbo, 2024-11-13:
Thank you very much for your answer, but we tried your method again and still encountered an error. Later, when we searched for the corresponding missing npy file, we found that the features and labels in the original decompressed dataset files did not correspond one-to-one. When we tried to delete the inconsistent files, we found that generate_training_stet.cy would run successfully. Therefore, we believe that it may be caused by incomplete data downloaded from Baidu Cloud. We hope you can check if the data set in the cloud drive is correct?

zhu-jinbo, 2024-11-14:
And I have another question, on your website `https://circuitnet.github.io/intro/intro.html` In this table, writing DRC in CircuitNet-N14 doesn't mean anything. Is it because CircuitNet-N14 doesn't use DRC?
[image]

apri0426, 2024-12-02:
> And I have another question, on your website `https://circuitnet.github.io/intro/intro.html` In this table, writing DRC in CircuitNet-N14 doesn't mean anything. Is it because CircuitNet-N14 doesn't use DRC? ![8eba51c0c3ee968754703652587587f](https://private-user-images.githubusercontent.com/179118182/386020421-ddab1bbe-5e86-4c21-98ca-bac0ffd8e402.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDgzOTgsIm5iZiI6MTczMzEwODA5OCwicGF0aCI6Ii8xNzkxMTgxODIvMzg2MDIwNDIxLWRkYWIxYmJlLTVlODYtNGMyMS05OGNhLWJhYzBmZmQ4ZTQwMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwMlQwMjU0NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYzNkYzE5YjQyNGFhMzRkZTNmOTliY2FjYjg3MzcyNDE5NmVkMGQ2ODljMzg1YTI2Y2NlOTNiODBmNzg2ODFjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.z3UJdnnmjvmn1tYLLzFPe66bHASi1sTKg7XN9NplXhs)

Yes, the DRC for N14 has potential problems, and it can't be fixed in the short term

closed 2025-03-02 by apri0426