circuitnet/CircuitNet #44: feature extraction

meihailang, 2024-11-11:
[image]
[image]
[image]

apri0426, 2024-11-12:
you need to first  run read_route_def to get gcell size then read_place_def

meihailang, 2024-11-12:
Thank you for your answer. I have another question, can you provide the corresponding netlist file?

梅海浪:-*
***@***.***

boice-girl, 2024-11-15:
I have a question,  congestion prediction is performed after the placement stage, so the features of macro_region, RUDY, and RUDY_pin should be extracted in the place_def, why do we need to read_rout_def first and obtain the gcell size of them? I compared the place def and the route def, and found that the gcell sizes and coordinates are different.

boice-girl, 2024-11-15:
> you need to first run read_route_def to get gcell size then read_place_def

I have a question, congestion prediction is performed after the placement stage, so the features of macro_region, RUDY, and RUDY_pin should be extracted in the place_def, why do we need to read_rout_def first and obtain the gcell size of them? I compared the place def and the route def, and found that the gcell sizes and coordinates are different.

apri0426, 2024-12-02:
> Thank you for your answer. I have another question, can you provide the corresponding netlist file? 梅海浪:-* ***@***.***  
> […](#)

Yes, the netlists are in the raw data dir in the netdisk.

boice-girl, 2024-12-11:
> > > you need to first run read_route_def to get gcell size then read_place_def
> > 
> > 
> > I have a question, congestion prediction is performed after the placement stage, so the features of macro_region, RUDY, and RUDY_pin should be extracted in the place_def, why do we need to read_rout_def first and obtain the gcell size of them? I compared the place def and the route def, and found that the gcell sizes and coordinates are different.
> 
> gcell is only used by EDA tool in global routing, and it is the basis for congestion analysis, so we use it as the grid in our flow. The gcell size is rather fixed for a specific technology, and you can also derive it from place_def with some more work. But we just obtain it from route_def directly, the other information from route_def is not used.

Thanks for your reply

closed 2025-03-02 by apri0426