circuitnet/CircuitNet #48: Requesting for sample congestion and drc reports for feature extraction

invincible-28, 2024-12-03:
Hello,

I am using CircuitNet to predict congestion and DRC, and have been successful in extracting the required input features from the lef and def files.

But, I am using Synopsys tools for my synthesis and backend flows, and thus am facing difficulties in extracting the labels from the congestion reports generated by Synopsys tools.

**Could you please provide sample reports that were used to extract these labels:**
'routability_features_decompressed/congestion/congestion_global_routing/overflow_based/congestion_GR_horizontal_overflow', 
 'routability_features_decompressed/congestion/congestion_global_routing/overflow_based/congestion_GR_vertical_overflow'
 'routability_features_decompressed/DRC/DRC_all'
'IR_drop_features_decompressed/IR_drop'
 
 **These are the reports that i am looking for:**
  self.parser.add_argument('--eGR_congestion_name', default='eGR_congestion') # congestion report from 'dumpNanoCongestArea'
  self.parser.add_argument('--route_congestion_name', default='route_congestion')
  self.parser.add_argument('--drc_rpt_name', default='drc.rpt')               # drc report from 'verify_drc'
  self.parser.add_argument('--twf_rpt_name', default='cts.twf')               # timing window file from 'write_timing_windows'
  self.parser.add_argument('--power_rpt_name', default='dyn_power.rpt')       # power report from 'report_power'
  self.parser.add_argument('--ir_rpt_name', default='route_dynamic_ir.rpt') 
  
**It would be easier for me to understand the feature extraction code if I can extract the labels from the reports that you used and then modify it accordingly for the Synopsys generated reports.**

P.S.: Thanks in advance for your reply and also for answering all my previous queries:)

apri0426, 2024-12-16:
You may find the sample in Innovus text command reference (TCR) which is a user guide that provides detailed explanations for each command. You can search the command listed above, like ''dumpNanoCongestArea'', and there is a output format.

invincible-28, 2024-12-19:
Yes, i found it and have been able to modify the code to extract the congestion labels from synopsys generated report. Thanks for the reply.

closed 2024-12-19 by invincible-28

zhu-jinbo, 2025-03-13:
> Yes, i found it and have been able to modify the code to extract the congestion labels from synopsys generated report. Thanks for the reply.

You have generated the corresponding DRC report. Can I ask you about the generation process? I want to learn