circuitnet/CircuitNet #50: Question about DRC Violations Prediction and RouteNet

wwwwwly, 2024-12-23:
Thank you very much for your work, which has been incredibly helpful to me. However, I encountered some questions while reading the paper **(_CircuitNet: An Open-Source Dataset for Machine Learning in VLSI CAD Applications With Improved Domain-Specific Evaluation Metric and Learning Strategies_)** and exploring the source code of the CircuitNet project.

1.  **Labels for Model Training**  
    In the paper, the task is described as: _Given a globally routed design, build a model to predict DRC violations after detailed routing. This problem is formulated as a DRC hotspot detection task, with a threshold to categorize DRC hotspot regions and non-hotspot regions._
The mathematical formulation is given as: 
$g_{DRC}: X \in \mathbb{R}^{w \times h \times f} \rightarrow V_i \in \\{0,1\\}^{w \times h}$
    However, upon examining the source code, I found that the labels used in the training process are continuous values, rather than binary (0-1) images. 
    
2.  **Loss Calculation**  
    In the source code of RouteNet model, the decoder's final layer uses a **sigmoid** activation function to map the model output to the range [0,1]. Meanwhile, the labels are normalized using **min-max** scaling. Is it appropriate to use **Mean Squared Error (MSE)** as the loss function in this case, given the difference in how the output and labels are processed?

apri0426, 2024-12-30:
1.  In this implementation, the label is binary only in inference. Of course, binary labels can be used for training, and the performance is better, but in this case, the threshold for the model becomes fixed.
2.  I don't understand your point. After min-max scaling, the labels are also in the range [0,1]

wwwwwly, 2025-02-17:
Thank you for your response. The previous issue has been temporarily resolved. However, when reading the feature extraction code (`read.py`), I encountered a few questions again. First, let me clarify that my major is Computer Science and Technology, so I don't have a deep understanding of digital integrated circuit back-end design. Therefore, the questions I ask might seem somewhat naive.

1. In `read.py` / `class ReadInnovusOutput` / `read_place_def()`, the regular expression `\d+` is used to obtain the bounding box of pins in the `place_pin_dict`, but it does not account for the negative sign (`-`). However, after examining the DEF file, I noticed that the coordinates can potentially be negative.

[image]

2. According to the **_LEF/DEF 5.7 Language Reference_**, the syntax for the `PINS` section of the DEF file is as follows: `+ LAYER layerName [SPACING minSpacing | DESIGNRULEWIDTH effectiveWidth] pt pt`, where the symbol `pt` is described as: _Represents a point in the design. This value corresponds to a coordinate pair, such as x y. You must enclose a point within parentheses, with space between the parentheses and the coordinates. For example, RECT ( 1000 2000 ) ( 1500 400 )._
Therefore, the value of `place_pin_dict` should be $[x_{\text{lower left}}, y_{\text{lower left}}, x_{\text{upper right}}, y_{\text{upper right}}]$. However, in `read.py` / `class ReadInnovusOutput` / `get_RUDY()`, the values are unpacked using `pin_left, pin_right, pin_lower, pin_upper = self.place_pin_dict[cell_pin_pair[1]]`.

apri0426, 2025-03-02:
Thank you for your report, I think you are right, the obtained pin shape is wrong. Since it only affects the "PINS" section in  the DEF, and the pin shape is much less important than pin position in this case, this issue doesn't make much difference to the resulted features.

wwwwwly, 2025-04-03:
Thank you for your response. I encountered another issue while reading `feature_extraction/src/read.py`, specifically regarding the extraction of the feature `egr_congestion`.  

In the function `read_eGR_congestion`, you derive a feature value array based on the coordinates from the **early global routing (eGR)** report. Then, you use the function `reshape()` (lines 420–423) to match the shape of the **GCells extracted from DEF file**. However, in practice, I found that the GCell partitioning used in the congestion report obtained via `dumpCongestArea -all` differs from the actual GCell partitioning used during chip routing. 
According to the **Innovus User Guide**, the explanation is as follows: 
[image]
As a result, when the program reaches the function `reshape()` , it naturally crashes. I'm not sure whether this issue is related to the specific design.

wwwwwly, 2025-04-03:
Besides, I also have some integrated circuit knowledge questions slightly related to your project, specifically regarding random chip layout generation. Actually, I referenced your computational methods to perform feature extraction on the `ispd_2015_contest_benchmark` chips, but used Innovus APIs instead of manual DEF file parsing. Below, I present my observations using the `mgc_edit_dist_a` chip as an example:

1. After importing the design via `loadLefFile` and `loadDefFile`, executing `verify_drc -limit 10000000` immediately yields **~300k** violations. This appears related to power planning issues, as `get_pg_pins` failed to retrieve macros' PG pins.
[image]

2. The above issue directly leads to **1.3M** violations after `detailRoute`, which unexpectedly helped address the insufficient positive samples problem([image]). Most violations are `metal_short` types, but I'm unsure how to disable this specific DRC rule, `verify_drc -check_only regular` didn't work.

3. Applying the same flow to `ispd_2019_test2` also resulted in **~100k** violations. However, after executing `verify_drc -check_only regular` post-`detailRoute` to filter power-related violations, only **dozens** remained—this seems more reasonable.

Have you encountered similar issues? I would greatly appreciate any solutions or suggestions you could share. Thank you in advance for your time and expertise.